ROBOTIS-GIT/ros2arduino #14: Segmentation fault

nnn112358, 2019-02-24:
I think ros2arduino is a wonderful project.
I thought trying it, but I will not move so I want you to follow me.

〇enviroment 
Ubuntu :16.04 x64
ROS2 :Crystal Clemmys
Micro-XRCE-DDS Agent  :git checkout a495c65faa964ddc068ac6e1249f17f5c9f92787
 

〇Symptom
Ubuntu:$ MicroXRCEAgent serial /dev/ttyACM0 115200
and 
Arduino:Serial publisher start
	[File] - [Examples] - [ros2arduino] - Serial - [publisher]

An error occurs
　$ MicroXRCEAgent serial /dev/ttyUSB0 115200
	Serial agent initialization... OK
	Enter 'q' for exit
	RTPS Participant matched 1.f.a.6e.e.61.0.0.0.0.0.0|0.0.1.c1
	Segmentation fault

OpusK, 2019-02-24:
Hi, @nnn112358,

Did you install ROS2 by binary package? Or did you install with source build?

nnn112358, 2019-02-24:
I use a Building ROS 2 on Linux.
https://index.ros.org/doc/ros2/Installation/Linux-Development-Setup/

By the way, when I used the latest version of Micro-XRCE-DDS-Agent, I solved it.

~~~~~~~~~~~~~~~~~~
$ git clone --recursive https://github.com/eProsima/Micro-XRCE-DDS-Agent/ 
$ mkdir build && cd build
$ cmake -DTHIRDPARTY=ON -DCONFIG_UDP_TRANSPORT_MTU=4096 -DCONFIG_SERIAL_TRANSPORT_MTU=4096 ..
$ sudo make install
$ sudo ldconfig /usr/local/lib/
~~~~~~~~~~~~~~~~~~

OpusK, 2019-02-25:
I'll check this again and update the document.

Thank you for your feedback :)

closed 2019-02-25 by OpusK