ROBOTIS-GIT/ros2arduino #17: Support for Ardunino Uno

mattions, 2019-03-14:
I saw that the Arduino DUE is supported, but not the UNO.

Will the support be added, or does already work?

Thanks!

OpusK, 2019-03-15:
Hi, @mattions 

Since the memory requirement of XRCE-DDS currently used is larger than RAM of Uno and Mega, it can not be supported.

If any vendor makes a lightweight protocol that can support DDS (if it makes it possible to run below 2K RAM), it seems to be able to support it.

mattions, 2019-03-15:
Hi @OpusK,
Yeah that makes sense. I expected that the new DDS stack was a bit heavier.

I guess at the moment nobody is working on something like that

OpusK, 2019-03-18:
> I guess at the moment nobody is working on something like that

Sadly, free DDS seems to be what you said.
Some paid DDS-RPC products require a lower source than MicroXRCE-DDS, but because they require a paid license, it is difficult to use in the Arduino project.

For now, for use with UNO, Mega, like rosserial, we need to use a separate package to convert and convert to DDS format. Unfortunately, ros2arduino does not yet have a specific plan associated with it.

closed 2019-03-25 by OpusK