ROBOTIS-GIT/ros2arduino #18: nav_msgs::Odometry can't use

quaei676, 2019-03-24:
I can use Pose or PoseWithCovariance to transfer the data.
But I dont't know why I can't use the nav_msgs::Odometry.
ros2arduino version:0.0.9
OpenCR version: 1.4.5
Many thanks.

OpusK, 2019-03-25:
I also saw this issue. We have found that data is not being transmitted after the writer is created.
In this regard, we will debug and answer your questions.

OpusK, 2019-07-02:
Hi, 

As Micro-XRCE-DDS has been updated, this issue seems to be resolved.
This update is currently applied to ros2arduino 0.1.1.

However, the following issues still exist and it is difficult to receive normal data.
If the issue below is resolved, the Odometry message seems to be available normally.

 - [Micro-XRCE-DDS-Agent#99](https://github.com/eProsima/Micro-XRCE-DDS-Agent/issues/99)

quaei676, 2019-07-02:
Hi, @OpusK 

Thanks for answering my questions. 
I will look forward to next version.
Many thanks.

OpusK, 2019-07-30:
@quaei676 

This issue is fixed in version [0.1.2](https://github.com/ROBOTIS-GIT/ros2arduino/releases/tag/0.1.2).
Please try after the update.

quaei676, 2019-08-01:
@OpusK 

My robot needs imu data.
Therefore, I directly use sensors.imu_.rpy[].
However, I find that I have to subscribe imu data, 
then sensors.imu_.rpy[] have normal data. 
Many thanks.

closed 2019-08-29 by OpusK