ROBOTIS-GIT/ros2arduino #19: Serial example on ESP32 does not start

asukiaaa, 2019-05-01:
I tried to run a serial publishing example on ESP32-WROOM-32(4MB Flash) but it loops rebooting.
Is there any additional configuration to run serial publishing on ESP32 or does not support this?

I got these error codes through serial port for ESP32.
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5860
entry 0x40080698
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d3928  PS      : 0x00060d30  A0      : 0x800d38a6  A1      : 0x3ffe39e0  
A2      : 0x3ffc006e  A3      : 0x00000029  A4      : 0x3ffc0d44  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x3ffe3b2c  A9      : 0x3ffe39e0  
A10     : 0x00000000  A11     : 0x3ffc006e  A12     : 0x00000029  A13     : 0x3ffe3b26  
A14     : 0x000000dd  A15     : 0x0000000f  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000  

Backtrace: 0x400d3928:0x3ffe39e0 0x400d38a3:0x3ffe3a00 0x400d3d0b:0x3ffe3a20 0x400d3447:0x3ffe3a60 0x400fb103:0x3ffe3aa0 0x400d2489:0x3ffe3ac0 0x400d253a:0x3ffe3ae0 0x400d1c20:0x3ffe3b50 0x400d16a8:0x3ffe3b70 0x400d1727:0x3ffe3b90 0x400d159f:0x3ffe3bb0 0x400d8603:0x3ffe3bd0 0x400826ab:0x3ffe3bf0 0x400828a0:0x3ffe3c20 0x40078f93:0x3ffe3c40 0x40078ff9:0x3ffe3c70 0x40079004:0x3ffe3ca0 0x400791a3:0x3ffe3cc0 0x400806ca:0x3ffe3df0 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

Rebooting...
ets Jun  8 2016 00:22:57
```

Thank you.

OpusK, 2019-05-08:
Hi, @asukiaaa 

I tested ESP32 based on wifi.
I will debug about Serial.

asukiaaa, 2019-05-08:
Hi @OpusK 

Thank you.
I wait for updating.

OpusK, 2019-05-15:
Hi, @asukiaaa 

In my ESP32, it works normally. (Not WROOM, I do not have this.)

Could you tell me the debugging information from Backtrace above?
If you use [ESP Exception Decoder](https://github.com/me-no-dev/EspExceptionDecoder), you can get backtrace details.

asukiaaa, 2019-05-18:
Hi @OpusK 

Thank you for checking but I cannot communicate with using ESP32-WROOM-32(4MB flash) and M5StackFire(16MB flash).

What I did to install Micro-XRCE-DDS-Agent.

```sh
git clone https://github.com/eProsima/Micro-XRCE-DDS-Agent.git
cd Micro-XRCE-DDS-Agent
git checkout b3c43bd20b16bd3570a128f05a9bf1a164883435
mkdir build && cd build
cmake -DTHIRDPARTY=ON -DCONFIG_UDP_TRANSPORT_MTU=4096 -DCONFIG_SERIAL_TRANSPORT_MTU=4096 ..
sudo make install
sudo ldconfig /usr/local/lib/
```

After install publishing program to ESP32 with using ArduinoIDE v1.8.9, I started Micro-XRCE-DDS-Agent with this command.

```sh
MicroXRCEAgent --serial /dev/ttyUSB0 115200
```

[image]
[image]

I set EspExceptionDecoder to ArduinoIDE and install program to ESP32 but nothing shown on stack trace window.
Could you give me some advice to trace the exception because I'm not familiar with this tool.

[image]

Thank you.

OpusK, 2019-06-03:
> Could you give me some advice to trace the exception?

Please see the link below for instructions.
 - [EspExceptionDecoder Usage](https://github.com/me-no-dev/EspExceptionDecoder#usage)

Did you press the ESP32 reset button while the Agent was running?

And, for debugging information, please add the `VERBOSE` option when building the Agent.

```
cmake -DTHIRDPARTY=ON -DVERBOSE=ON -DCONFIG_UDP_TRANSPORT_MTU=4096 -DCONFIG_SERIAL_TRANSPORT_MTU=4096 ..
sudo make install
sudo ldconfig /usr/local/lib/
```

asukiaaa, 2019-06-03:
Thank you for the information.

> Did you press the ESP32 reset button while the Agent was running?

No I did not.
I got the following logs when I reset ESP32 at running Micro-XRCE-DDS-Agent that activating VERBOSE option.
```
$ MicroXRCEAgent --serial /dev/ttyUSB0 115200
Serial agent initialization... OK
Enter 'q' for exit
<== [Client connected | session: 0x81 | client key: 0xAA 0xBB 0xCC 0xDD ]
<== [Create XRCE object | client key: 0xAA 0xBB 0xCC 0xDD | id: 0x0011 | PARTICIPANT]
<== [Create XRCE object | client key: 0xAA 0xBB 0xCC 0xDD | id: 0x0013 | PUBLISHER | participant id: 0x0011]
<== [Create XRCE object | client key: 0xAA 0xBB 0xCC 0xDD | id: 0x0014 | SUBSCRIBER | participant id: 0x0011]
<== [Create XRCE object | client key: 0xAA 0xBB 0xCC 0xDD | id: 0x0052 | TOPIC | participant id: 0x0011]
Segmentation fault
```

asukiaaa, 2019-06-03:
I got topic from ESP32 via Serial port!

[image]
[image]

I needed to activate ros2 before running `Micro-XRCE-DDS-Agent` and also needed to reset ESP after running `Micro-XRCE-DDS-Agent`.
Thank you very much sharing many information.

Please let me ask one more question before closing this issue.

Is is possible to communicate `ros2serial` and `Micro-XRCE-DDS-Agent` without resetting?
If not, it seems difficult that this libray replace `rosserial-arduino` because I need to reset usb device manually.

OpusK, 2019-06-04:
@asukiaaa ,

> I needed to activate ros2 before running Micro-XRCE-DDS-Agent

To use Agent with ROS2, you must use FastRTPS, which is used by ROS2. Therefore, Agent must be run after ROS2 first.

> Is it possible to communicate ros2serial and Micro-XRCE-DDS-Agent without resetting?

When you first connect to the Agent, you establish a session connection, which you can modify through the [configuration file](https://github.com/ROBOTIS-GIT/ros2arduino/blob/master/src/uxr/client/config.h#L46).
Currently, this value is minimal and we plan to increase the value in future updates. (when supporting dashing)

However, if the Agent does not exist until this count is exceeded, the Client (ros2arduino) will no longer try.

Anyway, if you need to test immediately, try modifying the two values below!
[`UXR_CONFIG_MAX_SESSION_CONNECTION_ATTEMPTS`](https://github.com/ROBOTIS-GIT/ros2arduino/blob/master/src/uxr/client/config.h#L46)
[`UXR_CONFIG_MIN_SESSION_CONNECTION_INTERVAL`](https://github.com/ROBOTIS-GIT/ros2arduino/blob/master/src/uxr/client/config.h#L47)

asukiaaa, 2019-06-04:
> Anyway, if you need to test immediately, try modifying the two values below!
> UXR_CONFIG_MAX_SESSION_CONNECTION_ATTEMPTS
> UXR_CONFIG_MIN_SESSION_CONNECTION_INTERVAL

The values of library on my PC were 10 and 1 so I changed both to 100.
After that ESP32 automatically communicate with `MicroXRCEAgent`.

Thank you very much for sharing information.
I'm looking forward to releasing next version.

closed 2019-06-04 by asukiaaa

grafoteka, 2020-12-14:
Hello, I'm having the same problem. 
I'm running Ubuntu 20.4, with ROS2 Foxy and using a ESP32 Module (https://www.diymore.cc/collections/esp8266/products/esp32-wifi-bluetooth-cp2104-dht11-soil-temperature-humidity-sensor-18650-battery-base) 

I have try the "publisher" example, but I cannot get any message in ROS2.

First, I run XRCE Agent

```
$ MicroXRCEAgent serial --dev /dev/ttyUSB0 -b 115200 -v 6
Press CTRL+C to exit
[1607931139.777002] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1607931139.777908] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
```
When I reset the ESP32 module, I got the next messages:
```
[1607931256.197901] info     | Root.cpp           | create_client            | create                 | client_key: 0xAABBCCDD, session_id: 0x81
[1607931256.198040] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x2864434397, address: 0
[1607931256.198155] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1607931256.201832] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA BB CC DD 81 00 FC 07
[1607931256.202024] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x2864434397, address: 0
[1607931256.202182] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1607931256.207282] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA BB CC DD 81 00 FC 07
[1607931256.207410] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x2864434397, address: 0
[1607931256.207523] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1607931256.271426] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 108, data: 
0000: 81 80 00 00 01 05 64 00 00 0A 00 11 01 02 00 00 55 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 72 6F 73 32 61 72 64 75 69 6E 6F 5F 70 75
0040: 62 5F 6E 6F 64 65 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63 69 70 61 6E
0060: 74 3E 3C 2F 64 64 73 3E 00 00 00 00
[1607931256.275972] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0xAABBCCDD, participant_id: 0x001(1)
[1607931256.276142] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 11 00 00
[1607931256.276254] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1607931256.283285] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1607931256.283434] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1607931256.288333] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1607931256.394226] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 24, data: 
0000: 81 80 01 00 01 05 0F 00 00 0B 00 13 03 02 00 00 01 00 00 00 00 00 11 00
[1607931256.394560] debug    | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0xAABBCCDD, publisher_id: 0x001(3), participant_id: 0x001(1)
[1607931256.394697] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 13 00 00
[1607931256.394717] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1607931256.403551] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1607931256.407289] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 24, data: 
0000: 81 80 02 00 01 05 0F 00 00 0C 00 14 04 02 00 00 01 00 00 00 00 00 11 00
[1607931256.407499] debug    | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0xAABBCCDD, subscriber_id: 0x001(4), participant_id: 0x001(1)
[1607931256.407627] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
[... cut ...]
[1607931256.551152] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 05 00 00 00 80
[1607931256.559236] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 05 00 00 00 80

```
But then, there is not any topic or node active and detected in ros2.

@asukiaaa how do you activate ros2 before running Micro-XRCE-DDS-Agent?

Kinds regards, 
Jorge

NicoZobernig, 2021-02-26:
Hi Jorge @grafoteka 
We are running into the same problem with the same setup. Where you able to solve this?
Thanks
Simon

grafoteka, 2021-03-03:
Hi @NicoZobernig I was not able  to solve this problem. Could you do any progress?
Thanks 
Jorge