ROBOTIS-GIT/ros2arduino #23: Micro XRCE-DDS Agent not working

giovanidiniz, 2019-06-16:
@OpusK 
Thank you for updating the instruction to Dashing. I now can execute all procedures without a problem on my TB3. The only issue I'm still having is with (I suspect) the Micro XRCE agent.
In the instructions, I can't check out (what I think is) the right version of the agent to link the OpenCR. This:
`git checkout 26d6ee42d9812a3465548590a66588d81c511ddc`
give me error "git reference is not a tree". Could this be because eProsima hasn't yet updated their repos?
Despite the error, I can use the Agent but my /cmd_vel messages are not received on the OpenCR. I know th erest of my setup works because it runs with the gazebo sim.
What is curious to me is that the /odom messages arrive on the "ROS" side of the setup but all values are rounded zeros, so I suspect they are not really being pushed from the OpenCR.
I'd be happy to help out getting this fixed. If you can point me in the right direction.
Thanks!

OpusK, 2019-06-17:
Hi, @giovanidiniz 

Have you checked ros2arduino's updated [README](https://github.com/ROBOTIS-GIT/ros2arduino#version-specific-dependencies)?

> Despite the error, I can use the Agent but my /cmd_vel messages are not received on the OpenCR. I know th erest of my setup works because it runs with the gazebo sim.

If you build git checkout again, it seems to be resolved.

> What is curious to me is that the /odom messages arrive on the "ROS" side of the setup but all values are rounded zeros, so I suspect they are not really being pushed from the OpenCR.

Currently, in the ROS2 package, /odom data is being sent from the Turtlebot3 package node, not OpenCR.
The Dashing package for TB3 has not yet been released, so it may be a related issue.

giovanidiniz, 2019-06-17:
Hi @OpusK.
I just tried to build it with the git check out command but I still get the same behavior 
`git checkout 26d6ee42d9812a3465548590a66588d81c511ddc` 
> fatal: reference is not a tree: 26d6ee42d9812a3465548590a66588d81c511ddc

Regarding the /odom, I didn't realize the OpenCR wasn't actually publishing the messages. Thank you for clarifying that. 
That is even weirder to me, however, because the gazebo sim is working just fine for me. Isn't the simulator also using the same  TB3 package node to generate the odom messages?

OpusK, 2019-06-17:
@giovanidiniz 

> git checkout 26d6ee42d9812a3465548590a66588d81c511ddc

In the updated README I linked above, it says `git checkout v1.1.0`. Would you check it again?

>  Isn't the simulator also using the same TB3 package node to generate the odom messages?

In this regard, it would be nice to leave it in the turtlebot3 repository issue. I do not know about Gazebo because I have not used it.

giovanidiniz, 2019-06-17:
Well, this is what I see in the README
```$ git clone https://github.com/eProsima/Micro-XRCE-DDS-Agent.git
$ cd Micro-XRCE-DDS-Agent && git checkout 26d6ee42d9812a3465548590a66588d81c511ddc
$ mkdir build && cd build
$ cmake ..
$ make
$ sudo make install
$ sudo ldconfig /usr/local/lib/
```
Anyways, I tried using git checkout v1.1.0 and that doesn't give errors. 
I still don't have the full topic list, e.g., cmd_vel, reset, motor_power, are no spawn when I run 
```ros2 launch turtlebot3_bringup robot.launch.py```

I then tried to install both client and agent using eProxifroma's instructions from [here](https://micro-xrce-dds.readthedocs.io/en/latest/installation.html). 

I'm at a loss now. Haven't dug dep into th eturtlebot3_node but I'm pretty sure it's not getting any info from the OpenCR. 

Any suggestions?

OpusK, 2019-06-18:
@giovanidiniz 

I changed the table in the README, but did not change the command description. I have modified it now.

> ros2 launch turtlebot3_bringup robot.launch.py

Currently, the TB3 package that supports ROS2 Dashing has not been released. Its behavior can not be guaranteed.

> I'm pretty sure it's not getting any info from the OpenCR. At least that is the only way I see odometry readings all 0.0 and not some very small numbers

I need more information. Please give me the information below.

- The TB3 package version (or commit)
- Agent command (+ log)
- OpenCR TB3 core firmware version (commit)

OpusK, 2019-07-30:
@giovanidiniz 

I have made an OpenCR firmware update, and have made an e-manual update.
I did not completely check everything I had to check, but I did update after the simple test.
Please refer to the [e-manual](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2).

This is a temporary update, and since most developers are currently on vacation, it is likely that developers will be able to begin detailed checks next week.

OpusK, 2019-08-29:
Hi, @giovanidiniz 

As for TB3, it was solved in a different way, as covered in ROBOTIS-GIT/turtlebot3#460.
Please refer to the new e-Manual.

closed 2019-09-03 by OpusK