ROBOTIS-GIT/ros2arduino #24: Missing topic for running client

yurumi, 2019-06-24:
Hi @OpusK ,

thank you very much for creating ros2arduino!

I have a problem getting it fully to work. I use an ESP32 NodeMCU. After finding a suitable USB port, I was able to flash the udp publisher example. The previously (on Arch Linux) started XRCE agent gave the following:
```
$ MicroXRCEAgent udp -p 2018
Enter 'q' for exit
[1561411294.860510] info     | UDPServerLinux.cpp | init                     | running...             | port: 2018
[1561411316.612634] info     | Root.cpp           | create_client            | create                 | client_key: 0xAABBCCDD, session_id: 0x81
[1561411316.612674] info     | UDPServerBase.cpp  | on_create_client         | session established    | client_key: 0xAABBCCDD, address: 192.168.0.215:47138
```

Also the node list is looking good:
```
$ ros2 node list
/ros2_xrcedds_participant
```

But the test topic seems to be missing:
```
$ ros2 topic list
/parameter_events
/rosout
```

Also something seems to be wrong with the node:
```
$ ros2 node info /ros2_xrcedds_participant 
/ros2_xrcedds_participant
[ERROR] [rmw_fastrtps_shared_cpp]: Unable to find GUID for node: ros2_xrcedds_participant
Failed to get_subscriber_names_and_types: Unable to find GUID for node , at /opt/ros2_dashing/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_node_info_and_types.cpp:85
```

Do you have any idea what could have gone wrong?

Cheers,
Thomas

OpusK, 2019-06-26:
Hi, @yurumi 

It seems to me that Agent has not been able to create other objects since client creation.
I need more information to know if the communication itself is missing or for some other reason.

> MicroXRCEAgent udp -p 2018

Would you like to redo the above command using the following command again?

> MicroXRCEAgent udp -p 2018 -v 6

yurumi, 2019-06-26:
Thanks for the quick reply! Here's the output of the above command:
```
MicroXRCEAgent udp -p 2018 -v 6 
Enter 'q' for exit
[1561583807.717600] info     | UDPServerLinux.cpp | init                     | running...             | port: 2018
[1561583807.717704] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1561583832.770515] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA BB CC DD 81 00 FC 07
[1561583832.771100] info     | Root.cpp           | create_client            | create                 | client_key: 0xAABBCCDD, session_id: 0x81
[1561583832.771145] info     | UDPServerBase.cpp  | on_create_client         | session established    | client_key: 0xAABBCCDD, address: 192.168.0.215:47138
[1561583832.771225] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1561583832.781035] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 112, data: 
0000: 81 80 00 00 01 05 68 00 00 0A 00 11 01 02 00 00 59 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 72 6F 73 32 5F 78 72 63 65 64 64 73 5F 70
0040: 61 72 74 69 63 69 70 61 6E 74 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63
0060: 69 70 61 6E 74 3E 3C 2F 64 64 73 3E 00 00 00 00
[1561583832.788279] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0xAABBCCDD, participant_id: 0x001(1)
[1561583832.788353] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 11 00 00
[1561583832.788380] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1561583832.788614] trace    | FastEntities.cpp   | onParticipantDiscovery   | matched                | entity_id: 0.0.1.c1, guid_prefix: 1.f.d0.bc.c3.a.0.0.1.0.0.0
[1561583832.799885] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1561583832.801511] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 24, data: 
0000: 81 80 01 00 01 05 0F 00 00 0B 00 14 04 02 00 00 01 00 00 00 00 00 11 73
[1561583832.801573] debug    | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0xAABBCCDD, subscriber_id: 0x001(4), participant_id: 0x001(1)
[1561583832.801620] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 14 00 00
[1561583832.801641] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1561583832.806031] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
```

To me this looks good (except the suspicious client key). Do you have any insights?

OpusK, 2019-06-28:
Hi, @yurumi 

I've found a bug related to this. We will release it with some other fixes.
If you want to work right now, change [STREAM_HISTORY](https://github.com/ROBOTIS-GIT/ros2arduino/blob/master/src/ros2/xrcedds/xrcedds.cpp#L22) to:
```
#define STREAM_HISTORY 2 //or above
```

yurumi, 2019-06-29:
Hi @OpusK,

that worked like a charm. Thank you very much!

Cheers,
Thomas

closed 2019-06-29 by yurumi