ROBOTIS-GIT/ros2arduino #30: on_create_client | session established | client_key: 0xAABBCCDD, address: ... Segmentation fault (core dumped)

yottayuan, 2019-10-06:
Ubuntu 18.04 LTS:
When I first run: MicroXRCEAgent udp -p 2018, the client connected well.
Then I close the Agent.
After I started the Agent and client again,
It shows: on_create_client | session established | client_key: 0xAABBCCDD, address: ... Segmentation fault (core dumped)
Have u meet this issue?

OpusK, 2019-10-07:
Hi, @yottayuan 

It seems to be an issue caused by releasing Ctrl + C and not releasing resources.

Would you like to leave an issue with [eProsima for Agent-related information](https://github.com/eProsima/Micro-XRCE-DDS-Agent)?
And, please exit as recommended by the vendor if possible.

yottayuan, 2019-10-12:
I found this issue: Not follow https://micro-xrce-dds.readthedocs.io/en/latest/installation.html#installing-the-agent-stand-alone
but follow https://micro-xrce-dds.readthedocs.io/en/latest/installation.html#installing-agent-and-client,
When I Install Agent stand-alone the issue disappeared.
Thanks OpusK :)

closed 2019-10-12 by yottayuan