ROBOTIS-GIT/ros2arduino #31: Questions about to support multiple topics.

asukiaaa, 2019-10-06:
I handle multiple topics (pub and sub) on a device with using rosserial-arduiono.
I want to handle like that on ROS2 but ros2arduino supports only one topic.
Please allow me to ask some questions about that.

I'm not familiar with Micro-XRCE-DDS but does the Micro-XRCE-DDS have limitation for one topic?
Do you plan to support multiple topics?
If I want to realize handling multiple topics (pub and sub) with using device info on ROS2, should I create a program of service that handle pub and sub with communicating with a device?

Thank you for sharing an useful project.

OpusK, 2019-10-07:
Hi, @asukiaaa 

I do not understand your words well.
I think ros2arduino can use multiple topics.

The link below is an example of using a number of topics from past TB3 cores.
Is this what you want?
 - [OpenCR TB3 core](https://github.com/ROBOTIS-GIT/OpenCR/blob/ba8c855af925e2a8d598534b59cad4baf80a9c10/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros2/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core_config.h#L173)

asukiaaa, 2019-10-07:
Hi @OpusK 

Thank you for the information.
I succeeded in creating pub and sub program and saw the topics `arduino_chatter` and `arduino_listener` by `ros2 topic list`.
I miss understood about `Only one node available` on readme.

However, I execute
```
RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 topic echo /arduino_chatter
```
but console prints
```
Could not determine the type for the passed topic
```
(publisher example also causes this problem.)

Do you know how to avoid this problem?

My environment
OS: Ubuntu18.04
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.2 LTS
Release:	18.04
Codename:	bionic
```

ROS: dashing
```
$ apt-cache show ros-dashing-ros-base 
Package: ros-dashing-ros-base
Version: 0.7.2-1bionic.20190923.212341
Architecture: amd64
```

Micro-XRCE-DDS Agent: v1.1.0

Thank you.

OpusK, 2019-10-08:
@asukiaaa 

> RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 topic echo /arduino_chatter

The agent used by ros2arduino is eProsima's Micro-XRCE-DDS.
This is only available with eProsima's FastRTPS.

asukiaaa, 2019-10-08:
@OpusK 

Thank you for the information.
My Ubuntu did not allow me to install `ros-dashing-fastrtps` because of breaking dependency about libssl so I install Ubuntu18.04.3 and suceeded in installing `ros-dashing-fastrtps`.

I succeeded in running MicroXRCEAgent on a terminal which does not load ros2. (If I run it on terminal which loaded ros2, that shows segmentation fault.)
```sh
$ MicroXRCEAgent serial --dev /dev/ttyUSB0 -b 115200
Enter 'q' for exit
[1570542701.897215] info     | SerialServerLinux.cpp | init                     | running...             | fd: 3
[1570542705.695600] info     | Root.cpp           | create_client            | create                 | client_key: 0xAABBCCDD, session_id: 0x81
[1570542705.695719] info     | SerialServerBase.cpp | on_create_client         | session established    | client_key: 0xAABBCCDD, address: 0
```

However, I cannot find topic by ros2.
```sh
$ ros2 topic list
/parameter_events
/rosout
```

Could you give me additional advice?

OpusK, 2019-10-10:
@asukiaaa 

You need to use agent with ros2 loaded. (Or, the FastRTPS library you use with ros2 must be the same)
However, we need to find out why the segmentation fault occurred. First, we can see more detailed debugging messages through the [`-v option`](https://micro-xrce-dds.readthedocs.io/en/latest/agent.html#run-an-agent).

Please add -v 6 to check the debug messages.
And, I think it's good to raise this debugging information in [eProsima's repository](https://github.com/eProsima/Micro-XRCE-DDS-Agent/issues).

asukiaaa, 2019-10-13:
@OpusK 

Thank you for the information.
I could see more info by adding option.

```
MicroXRCEAgent serial --dev /dev/ttyUSB0 -b 115200 -v 6
Enter 'q' for exit
[1570970989.371558] info     | SerialServerLinux.cpp | init                     | running...             | fd: 3
[1570970989.371738] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1570970993.893798] debug    | SerialServerLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x00000000, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 04 00 05 00 80
[1570970996.010872] debug    | SerialServerLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA BB CC DD 81 00 FC 07
[1570970996.011148] info     | Root.cpp           | create_client            | create                 | client_key: 0xAABBCCDD, session_id: 0x81
[1570970996.011235] info     | SerialServerBase.cpp | on_create_client         | session established    | client_key: 0xAABBCCDD, address: 0
[1570970996.011381] debug    | SerialServerLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1570970996.029973] debug    | SerialServerLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 116, data: 
0000: 81 80 00 00 01 05 6C 00 00 0A 00 11 01 02 00 00 5D 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 72 6F 73 32 61 72 64 75 69 6E 6F 5F 70 75
0040: 62 5F 61 6E 64 5F 73 75 62 5F 6E 6F 64 65 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61
0060: 72 74 69 63 69 70 61 6E 74 3E 3C 2F 64 64 73 3E 00 00 00 00
[1570970996.033507] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0xAABBCCDD, participant_id: 0x001(1)
[1570970996.033603] debug    | SerialServerLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 11 00 00
[1570970996.033674] debug    | SerialServerLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1570970996.039625] debug    | SerialServerLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1570970996.044492] debug    | SerialServerLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 24, data: 
0000: 81 80 01 00 01 05 0F 00 00 0B 00 13 03 02 00 00 01 00 00 00 00 00 11 00
[1570970996.044579] debug    | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0xAABBCCDD, publisher_id: 0x001(3), participant_id: 0x001(1)
[1570970996.044638] debug    | SerialServerLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 13 00 00
[1570970996.044656] debug    | SerialServerLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1570970996.050673] debug    | SerialServerLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1570970996.055451] debug    | SerialServerLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 24, data: 
0000: 81 80 02 00 01 05 0F 00 00 0C 00 14 04 02 00 00 01 00 00 00 00 00 11 00
[1570970996.055537] debug    | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0xAABBCCDD, subscriber_id: 0x001(4), participant_id: 0x001(1)
[1570970996.055600] debug    | SerialServerLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 02 00 05 01 06 00 00 0C 00 14 00 00
[1570970996.055616] debug    | SerialServerLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1570970996.061603] debug    | SerialServerLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1570970996.159098] debug    | SerialServerLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 120, data: 
0000: 81 80 03 00 01 03 6D 00 00 0D 00 52 02 02 00 00 5F 00 00 00 3C 64 64 73 3E 3C 74 6F 70 69 63 3E
0020: 3C 6E 61 6D 65 3E 53 74 72 69 6E 67 3C 2F 6E 61 6D 65 3E 3C 64 61 74 61 54 79 70 65 3E 73 74 64
0040: 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 53 74 72 69 6E 67 5F 3C 2F 64 61 74 61 54
0060: 79 70 65 3E 3C 2F 74 6F 70 69 63 3E 3C 2F 64 64 73 3E 00 00 11 00 00 00
[1570970996.159524] debug    | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xAABBCCDD, topic_id: 0x005(2), participant_id: 0x001(1)
[1570970996.159616] debug    | SerialServerLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 03 00 05 01 06 00 00 0D 00 52 00 00
[... cut ...]
00A0: 72 69 74 65 72 3E 3C 2F 64 64 73 3E 00 00 13 00
Segmentation fault
```

> And, I think it's good to raise this debugging information in eProsima's repository.

Thank you for the suggestion but it is difficult for me to distinguish the problem occurs on ros2arduino or Micro-XRCE-DDS-Agent.
I will try that if I can get more motivation to solve this problem.
 
Anyway, my question about `Does ros2arduino support multiple topics?` was solved by answer YES so I close this issue.

Thank you for sharing information.

closed 2019-10-13 by asukiaaa

zeta0707, 2020-03-21:
Adding LD_PRELOAD="/usr/local/lib/libfastrtps.so.1" solved this problem.
zeta@changwhan:~/Downloads$ LD_PRELOAD="/usr/local/lib/libfastrtps.so.1" MicroXRCEAgent serial --dev /dev/ttyUSB0 -b 115200 -v 5

zeta@changwhan:~/Downloads$ MicroXRCEAgent serial --dev /dev/ttyUSB0 -b 115200 -v 6
Enter 'q' for exit
[1584787118.540571] info     | SerialServerLinux.cpp | init                     | running...             | fd: 3
[1584787118.540731] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1584787123.957046] debug    | SerialServerLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x00000000, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 10 00 13 00 80
[1584787127.177679] debug    | SerialServerLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA BB CC DD 81 00 FC 07
[1584787127.177911] info     | Root.cpp           | create_client            | create                 | client_key: 0xAABBCCDD, session_id: 0x81
[1584787127.178012] info     | SerialServerBase.cpp | on_create_client         | session established    | client_key: 0xAABBCCDD, address: 0
[1584787127.178197] debug    | SerialServerLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1584787127.181939] debug    | SerialServerLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA BB CC DD 81 00 FC 07
[1584787127.182224] debug    | SerialServerLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1584787127.193861] debug    | SerialServerLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 108, data: 
0000: 81 80 00 00 01 05 64 00 00 0A 00 11 01 02 00 00 55 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 72 6F 73 32 61 72 64 75 69 6E 6F 5F 70 75
0040: 62 5F 6E 6F 64 65 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63 69 70 61 6E
0060: 74 3E 3C 2F 64 64 73 3E 00 00 00 00
[1584787127.201007] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0xAABBCCDD, participant_id: 0x001(1)
[1584787127.201188] debug    | SerialServerLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 11 00 00
[1584787127.201240] debug    | SerialServerLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1584787127.206368] debug    | SerialServerLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1584787127.209798] debug    | SerialServerLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 24, data: 
0000: 81 80 01 00 01 05 0F 00 00 0B 00 13 03 02 00 00 01 00 00 00 00 00 11 00
[1584787127.209967] debug    | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0xAABBCCDD, publisher_id: 0x001(3), participant_id: 0x001(1)
[1584787127.210135] debug    | SerialServerLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 13 00 00
[1584787127.210179] debug    | SerialServerLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1584787127.215160] debug    | SerialServerLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1584787127.218759] debug    | SerialServerLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 24, data: 
0000: 81 80 02 00 01 05 0F 00 00 0C 00 14 04 02 00 00 01 00 00 00 00 00 11 00
[1584787127.218969] debug    | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0xAABBCCDD, subscriber_id: 0x001(4), participant_id: 0x001(1)
[1584787127.219200] debug    | SerialServerLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 02 00 05 01 06 00 00 0C 00 14 00 00
[1584787127.219258] debug    | SerialServerLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1584787127.224433] debug    | SerialServerLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1584787127.236507] debug    | SerialServerLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 120, data: 
0000: 81 80 03 00 01 03 6D 00 00 0D 00 52 02 02 00 00 5F 00 00 00 3C 64 64 73 3E 3C 74 6F 70 69 63 3E
0020: 3C 6E 61 6D 65 3E 53 74 72 69 6E 67 3C 2F 6E 61 6D 65 3E 3C 64 61 74 61 54 79 70 65 3E 73 74 64
[... cut ...]
[1584787188.219906] debug    | SerialServerLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 0A 00 00 00 80
[1584787189.219710] debug    | SerialServerLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 36, data: 
0000: 81 80 0A 00 07 01 1C 00 00 14 00 15 14 00 00 00 48 65 6C 6C 6F 20 72 6F 73 32 61 72 64 75 69 6E
0020: 6F 20 35 00
[1584787189.219895] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000001, len: 24, data: 
``

asukiaaa, 2020-03-21:
Do you compile Micro-XRCE-DDS-Agent with sharing dependencies of ROS2?

I got error when using different dependencies between ROS2 and Micro-XRCE-DDS-Agent.

```sh
git clone https://github.com/eProsima/Micro-XRCE-DDS-Agent.git
cd Micro-XRCE-DDS-Agent && git checkout v1.1.0
mkdir build && cd build
source /opt/ros/dashing/setup.bash # <<<<<< This command is important.
cmake ..
make
sudo make install
sudo ldconfig /usr/local/lib/
```

Referenc: My blog (in Japanese)
http://asukiaaa.blogspot.com/2020/01/ros2arduinomicroxrceagent.html

zeta0707, 2020-03-21:
Hi Asukiaaa,

Actually I found LD_PRELOAD workaround from your blog. Many thanks!
"source /opt/ros/dashing/setup.bash" is included in my .bashrc.
So the command is executed whenever terminal starts.
Though I faced the "Segmentation fault (core dumped)" if I don't use  LD_PRELOAD workaround.

zeta0707, 2020-03-22:
Hi Asukiaaa,

Now I solved the problem.
In my .bashrc, there were two lines.

source /opt/ros/dashing/setup.bash
source ~/turtlebot3_ws/install/setup.bash

I removed "~/turtlebot3_ws/install/setup.bash" from .bashrc and build Micro-XRCE-DDS-Agent.
I don't need LD_PRELOAD workaround anymore.

Thanks!

asukiaaa, 2020-03-23:
I'm glad to know that my blog post helped you and you overcome the problem.