ROBOTIS-GIT/ros2arduino #33: Could start the node before the Agent?

yottayuan, 2019-10-22:
Sorry for my poor English. Now it works well. I still have a question,maybe an enhancement.
Now , I must start the Agent first ,then start my Esp32 board. The node in the Esp32 board will connect the Agent. But when the Agent crashes, The board will not reconnect.
My question is : How to start the Agent after the Esp32 board started? Now I must reboot the Esp32 board. But n a production environment, It's not a good idea.

OpusK, 2019-11-06:
Hi, @yottayuan 

> My question is : How to start the Agent after the Esp32 board started? Now I must reboot the Esp32 board. But in a production environment, It's not a good idea.

Implementing this feature may require a redesign of ros2arduino, and most of all we should check with Agents and Clients (eProsima's products) to see if this is possible.
This seems to require time, and unfortunately, my current tasks make it difficult to get started for the time being.

OpusK, 2020-02-07:
Hi, @yottayuan 

Sorry, my answer is too late...
The link below is the code I used to create an example of XELNetwork CommXEL-W (using esp32).

Please apply and test the recreate () function.
- [XELNetwork Master example code link](https://github.com/ROBOTIS-GIT/XELNetworkMaster/blob/b04b9f243de8864ed770a4e172fe8db62e28598e/src/XELNetworkMaster.cpp#L135)

closed 2020-03-23 by OpusK