ROBOTIS-GIT/ros2arduino #34: Node constructor failing?

rnvandemark, 2019-11-05:
**Environment**
Ubuntu 18.04.3 LTS
ROS2 Dashing
Micro-XRCE-DDS-Agent v1.1.0
Arduino Due

**Problem**
I'm not sure how to debug further, but something is preventing the ros2::Node class from being instantiated properly. When I was first playing around with this a couple of months ago, the code below (or something very closely resembling it) was working correctly (meaning an analog voltage was being written to the pin successfully, and the DDS agent could be brought up and down without a problem). I've also tried compiling and uploading the code with the subscription instantiation commented out, and it still fails the same way. The first block of DEBUG LED code `loop` can be seen after uploading the sketch, but the second block cannot. When starting the Micro XRCE DDS Agent, the client is not created, it instead hangs after init, like so:

```
~$ MicroXRCEAgent serial --dev /dev/ttyACM0 -b 115200
Enter 'q' for exit
[1572982921.122000] info     | SerialServerLinux.cpp | init                     | running...             | fd: 3
```

After entering 'q', I get:

```
[1572983745.971604] info     | SerialServerLinux.cpp | close                    | server stopped         | fd: 3
Segmentation fault (core dumped)
```

```
#include <ros2arduino.h>

#define XRCEDDS_PORT Serial
#define PROGRAM_BAUD 115200

#define INTENSITY_LED 7

#define ANALOG_RESOLUTION_BITS 12

const static int ANALOG_NUM_VALUES = pow(2, ANALOG_RESOLUTION_BITS);

void intensity_change_callback(std_msgs::Float32* msg, void* arg)
{
  (void)(arg);
  
  int fixed_value = (int)(msg->data * ANALOG_NUM_VALUES);
  if (fixed_value >= ANALOG_NUM_VALUES)
  {
    fixed_value = ANALOG_NUM_VALUES - 1;
  }
  else if (fixed_value < 0)
  {
    fixed_value = 0;
  }
  
  analogWrite(INTENSITY_LED, fixed_value);
  digitalWrite(LED_BUILTIN, 1 - digitalRead(LED_BUILTIN));
}

class DueNode : public ros2::Node
{
//private:
//  ros2::Publisher<std_msgs::String>* due_debug_pub;

public:
  DueNode() : Node("due_node")
  {
    this->createSubscriber<std_msgs::Float32>(
      "smart_home/intensity_change_chatter",
      (ros2::CallbackFunc)intensity_change_callback,
      nullptr
    );
    
    //due_debug_pub =
    //  this->createPublisher<std_msgs::String>("smart_home/due_debug_chatter");
  }
};

void setup()
{
  XRCEDDS_PORT.begin(PROGRAM_BAUD);
  while (!XRCEDDS_PORT);
  
  pinMode(INTENSITY_LED, OUTPUT);
  pinMode(LED_BUILTIN, OUTPUT);
  analogWriteResolution(ANALOG_RESOLUTION_BITS);
  
  ros2::init(&XRCEDDS_PORT);
}

void loop()
{
  // DEBUG LED No. 1
  digitalWrite(LED_BUILTIN, HIGH);
  delay(1500);
  digitalWrite(LED_BUILTIN, LOW);
  delay(1500);
  
  static DueNode due_node;
  
  // DEBUG LED No. 2
  for (int i = 0; i < 5; i++)
  {
    digitalWrite(LED_BUILTIN, HIGH);
    delay(100);
    digitalWrite(LED_BUILTIN, LOW);
    delay(100);
  }
  
  ros2::spin(&due_node);
}
```

Any ideas on what's happening?

OpusK, 2019-11-06:
Hi, @rnvandemark 

```cpp
  XRCEDDS_PORT.begin(PROGRAM_BAUD);
  while (!XRCEDDS_PORT);`
```

When waiting with the code above, you need to run the agent.
Have you tested the basic subscribe example?

rnvandemark, 2019-11-06:
> Hi, @rnvandemark
> 
> ```c++
>   XRCEDDS_PORT.begin(PROGRAM_BAUD);
>   while (!XRCEDDS_PORT);`
> ```
> 
> When waiting with the code above, you need to run the agent.
> Have you tested the basic subscribe example?

Thanks for getting back to me! I just uploaded the subscriber example sketch and ran the agent again with the `MicroXRCEAgent serial --dev /dev/ttyACM0 -b 115200` command, and I got the same result (blocking on init, seg fault on quitting.)

OpusK, 2019-11-11:
@rnvandemark, I'm sorry for the late reply.

Is your ROS2 binary version? Or is it the source? Can you tell me the patch version?
Also, can you tell me how you installed Agent v1.1.0?

rnvandemark, 2019-11-11:
1. I installed ROS2 via the Debian packages.
2. According to `apt-cache`, I believe I have [Patch 4](https://github.com/ros2/ros2/tree/release-dashing-20191018):

```
Package: ros-dashing-desktop
Version: 0.7.2-1bionic.20191017.214317
etc...
```

(If there's a more standard and more definitive way to get it, then lemme know and I can get back to you.)

3. I did the standalone installation from [here](https://micro-xrce-dds.readthedocs.io/en/latest/installation.html#installing-the-agent-stand-alone), and I'm on Linux so I obviously did the Linux portion of it.

I could see the patch being the problem, because like I said I was running the agent and the code from the initial post above a couple of months ago successfully. Patch 3 and 4 have come out since the last time I ran it I think.

rnvandemark, 2019-11-12:
Actually, it doesn't seem this is solely a ros2arduino problem. I just ran the Agent by itself:

```
rnvandemark@ideapad: ~/$ MicroXRCEAgent 
A subcommand is required
Run with --help for more information.

Enter command: q
Segmentation fault (core dumped)
```

Just running the agent by itself and immediately quitting results in a segfault, and I tested with a fresh install of v1.1.0 to make sure... Although it is interesting that the Arduino sketch never gets past the ros2::Node constructor...

OpusK, 2019-11-18:
Hi, @rnvandemark 

> I could see the patch being the problem, because like I said I was running the agent and the code from the initial post above a couple of months ago successfully. Patch 3 and 4 have come out since the last time I ran it I think.

In fact, the version relationship between FastRTPS and MicroXRCEDDS is important.
I did not make any further updates after the Patch3 update.
Soon, I will test and update to the latest update. Thank you for your feedback.

jguggenh, 2019-12-24:
Hi @OpusK any update on this? I am probably having the same issue as @rnvandemark .

OpusK, 2020-02-07:
Hi, @jguggenh 

Did you solve this problem?

jguggenh, 2020-02-07:
Nope I did not. Just wrote a ros node that talks over serial to the arduino. I haven't look at it since around when I sent this message.

rnvandemark, 2020-02-15:
I just happened to try this on a fresh install of everything, including OS, and I'm getting a slightly different error, but still critical failure:

```
u18@u18-vb:~$ MicroXRCEAgent serial --dev /dev/ttyACM0 -b 115200
Enter 'q' for exit
[1581783142.208422] info     | SerialServerLinux.cpp | init                     | running...             | fd: 3
[1581783142.464702] info     | Root.cpp           | create_client            | create                 | client_key: 0xAABBCCDD, session_id: 0x81
[1581783142.464808] info     | SerialServerBase.cpp | on_create_client         | session established    | client_key: 0xAABBCCDD, address: 0
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
```

I made sure to follow the updated guidelines for software versions:

ros2arduino: 0.1.4
ROS2: >= Dashing Diademata Patch3 (I did the release version, so probably Patch4)
Micro-XRCE-DDS Agent >= 1.1.0 (I did 1.1.0)

rnvandemark, 2020-02-17:
So I tried to replicate the environment I had when this was functional for me, and I was able to create a working system, as far as I can tell... I uploaded the example ros2arduino publisher program to a Due and can see the arduino_chatter topic on my computer.

The environment that worked for me:

- Ubuntu 18.04.4
- ROS2 Dashing built from source (can use the instructions [here](https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Development-Setup/), just be sure to use a tag for a release that works when you reach the `wget` command, I used [release-dashing-20190806](https://github.com/ros2/ros2/blob/release-dashing-20190806/ros2.repos)'s)
- Micro-XRCE-DDS Agent 1.1.0
- ros2arduino 0.1.2

OpusK, 2020-02-18:
@rnvandemark, thanks for your feedback :)

 Like your feedback, it seems to be due to mismatching between ROS2 version and Agent, Client (ros2arduino) version. We need to find the right version for each.

However, since I currently do not have any resources to manage ros2arduino, I think I should take some personal time. Therefore, it is difficult to tell how long it will take.

zeta0707, 2020-03-21:
I'm not able to see /arduino_chatter somehow.

My environment
=============================
Ubuntu 18.04.4
ROS2 Dashing: install by ros2-dashing-20190910-linux-bionic-amd64.tar.bz2, patch level 3
Micro-XRCE-DDS Agent 1.1.0
ros2arduino 0.1.4
ESP8266 NodeMCU

Topic doesn't show /arduino_chatter, and node is blank.
=============================
zeta@changwhan:~$ ros2 topic list
/parameter_events
/rosout

zeta@changwhan:~$ ros2 node list

Any experience you see this kind of log?
=============================
zeta@changwhan:~/Micro-XRCE-DDS-Agent/build$ MicroXRCEAgent udp -p 2018 -v 6
Enter 'q' for exit
[1584778434.363903] info     | UDPServerLinux.cpp | init                     | running...             | port: 2018
[1584778434.364040] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1584778444.670083] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA BB CC DD 81 00 FC 07
[1584778444.670355] info     | Root.cpp           | create_client            | create                 | client_key: 0xAABBCCDD, session_id: 0x81
[1584778444.670456] info     | UDPServerBase.cpp  | on_create_client         | session established    | client_key: 0xAABBCCDD, address: 192.168.35.93:47138
[1584778444.670655] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1584778444.670863] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA BB CC DD 81 00 FC 07
[1584778444.671157] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1584778444.671853] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA BB CC DD 81 00 FC 07
[1584778444.672116] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00

I used below Arduino sketch.
=============================
#include <ros2arduino.h>
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

#define SSID       "XXXXXXXXXXX"
#define SSID_PW    "XXXXXXXXXXX"
#define AGENT_IP   "192.168.35.201"
#define AGENT_PORT 2018 //AGENT port number

#define PUBLISH_FREQUENCY 2 //hz

void publishString(std_msgs::String* msg, void* arg)
{
  (void)(arg);

  static int cnt = 0;
  sprintf(msg->data, "Hello ros2arduino %d", cnt++);
}

class StringPub : public ros2::Node
{
public:
  StringPub()
  : Node("ros2arduino_pub_node")
  {
    ros2::Publisher<std_msgs::String>* publisher_ = this->createPublisher<std_msgs::String>("arduino_chatter");
    this->createWallFreq(PUBLISH_FREQUENCY, (ros2::CallbackFunc)publishString, nullptr, publisher_);
  }
};

WiFiUDP udp;

void setup() 
{

  Serial.begin(115200);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(SSID);
  pinMode(LED_BUILTIN, OUTPUT); 
  
  WiFi.begin(SSID, SSID_PW);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  ros2::init(&udp, AGENT_IP, AGENT_PORT);
}

void loop() 
{
  static StringPub StringNode;

  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(100);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(100);                       // wait for a second 
  
  ros2::spin(&StringNode);
}

OpusK, 2020-03-23:
> Any experience you see this kind of log?

If you're curious about more about Log, it would be a good idea to ask eProsima.

In my view, there is only log for client creation, and there is no log for creation for participant, publisher, and data_writer, so it seems that participant creation was not successful. (I think it would be good to check where the communication stops using wireshark.)

zeta0707, 2020-03-27:
HI Opusk,

My wild guess is that NodeMCU can't be used, it is ESP12.
When I used Arduino DUE, "MicroXRCEAgent udp -p 2018" was fine.

**My environment**
Ubuntu 18.04.4
ROS2 Dashing: install by ros2-dashing-20190910-linux-bionic-amd64.tar.bz2, patch level 3
Micro-XRCE-DDS Agent 1.1.0
ros2arduino 0.1.4
ESP8266 NodeMCU -> "Arduino DUE + Ethernet shield"

Thanks!

OpusK, 2020-03-30:
Hi, @zeta0707 

Thanks for your feedback.
I will check the ESP8266 later when I have time.

DylanLN, 2020-04-11:
Hi,
@OpusK 
My environment
Ubuntu 18.04
ROS2 Dashing:
Micro-XRCE-DDS Agent 1.1.0
ros2arduino 0.1.4
ESP32 NodeMCU

I run the agent and then start esp32 will report the following error：
```bash
ln@ln-pctogo:~$ MicroXRCEAgent udp -p 2018
Enter 'q' for exit
[1586617850.209044] info     | UDPServerLinux.cpp | init                     | running...             | port: 2018
[1586617855.149720] info     | Root.cpp           | create_client            | create                 | client_key: 0xAABBCCDD, session_id: 0x81
[1586617855.149812] info     | UDPServerBase.cpp  | on_create_client         | session established    | client_key: 0xAABBCCDD, address: 192.168.0.104:47138
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
已放弃 (核心已转储)
```
may I know what is the reason

zeta0707, 2020-04-12:
Hi DylanLN,

I also got a same problem with following steps.
```

$ git clone https://github.com/eProsima/Micro-XRCE-DDS-Agent.git
$ cd Micro-XRCE-DDS-Agent && git checkout v1.1.0
$ mkdir build && cd build
$ source /opt/ros/dashing/setup.bash # to share libraries with ros2
$ cmake ..
$ make
$ sudo make install
$ sudo ldconfig /usr/local/lib/
```

Please try again with skipping "git checkout v1.1.0".
Or you can use "git checkout v.1.1.4" or higher.
There's fastrtps version mismatch, since ros2 are using higher version.

```
$ git clone https://github.com/eProsima/Micro-XRCE-DDS-Agent.git
$ cd Micro-XRCE-DDS-Agent
$ mkdir build && cd build
$ source /opt/ros/dashing/setup.bash # to share libraries with ros2
$ cmake ..
$ make
$ sudo make install
$ sudo ldconfig /usr/local/lib/

```
You can refer to https://blog.naver.com/zeta0807/221881999902, but it's Korean.

Thanks!

DylanLN, 2020-04-12:
Hi!
@zeta0707  
Thank you for helping me solve the problem.
 I tried according to your method and it turned out that the problem was solved, but I got stuck with the same problem as you.
No matter what i run:
```bash
ln@ln-pctogo:~$ MicroXRCEAgent udp -p 2018
Enter 'q' for exit
[1586672203.330924] info     | UDPServerLinux.cpp | init                     | running...             | port: 2018
[1586672212.952960] info     | Root.cpp           | create_client            | create                 | client_key: 0xAABBCCDD, session_id: 0x81
[1586672212.953073] info     | UDPServerBase.cpp  | on_create_client         | session established    | client_key: 0xAABBCCDD, address: 192.168.0.104:47138
```
Still running:
```bash
ln@ln-pctogo:~$ MicroXRCEAgent udp -p 2018 -v 6
Enter 'q' for exit
[1586672497.365896] info     | UDPServerLinux.cpp | init                     | running...             | port: 2018
[1586672497.366057] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1586672499.626483] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 EA 15 ED 15 80
[1586672505.106920] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA BB CC DD 81 00 FC 07
[1586672505.107224] info     | Root.cpp           | create_client            | create                 | client_key: 0xAABBCCDD, session_id: 0x81
[1586672505.107305] info     | UDPServerBase.cpp  | on_create_client         | session established    | client_key: 0xAABBCCDD, address: 192.168.0.104:47138
[1586672505.107428] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA BB CC DD 81 00 FC 07
[1586672505.107448] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1586672505.107512] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA BB CC DD 81 00 FC 07
[1586672505.107673] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1586672505.107721] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1586672505.121803] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA BB CC DD 81 00 FC 07
[1586672505.122043] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00

```
Will not receive messages from esp32:
```bash
ln@ln-pctogo:~$ ros2 topic list 
/parameter_events
/rosout
ln@ln-pctogo:~$ 
```
/(ㄒoㄒ)/~~

zeta0707, 2020-04-12:
I suspect you don't use esp32 module. Actually I faced the same problem
with Nodemcu, then i realized mine is esp12 module. Try again with esp32
based module.

OpusK, 2020-05-14:
Hi, all.

Version 0.2.0 has been released. (Tested with Dashing Patch 6(latest))
For more information, please refer to the README.

And, I am sorry about the ESP8266. 
I have tested it myself and found that there was a problem processing the XRCE-DDS protocol. 
I tested it by applying a few correction points, but the system crashed.

Perhaps this will be considered later.
Therefore, we removed ESP8266 from the UDP support list.
We apologize for the loss of your precious time.

closed 2020-05-27 by OpusK