ROBOTIS-GIT/ros2arduino #43: ros2 node info error

munakano, 2020-05-11:
Hi.

Thanks for creating ros2arduino.
I'm using the publisher sample code.
I have a problem getting node info.

**Environment**
Ubuntu 18.04.4 LTS
ROS2 Dashing
Micro-XRCE-DDS-Agent v1.1.4

**Probrem**
```
$ ros2 node info /ros2arduino_pub_node
/ros2arduino_pub_node
[ERROR] [rmw_fastrtps_shared_cpp]: Unable to find GUID for node: ros2arduino_pub_node
Failed to get_subscriber_names_and_types: Unable to find GUID for node , at /tmp/binarydeb/ros-dashing-rmw-fastrtps-shared-cpp-0.7.6/src/rmw_node_info_and_types.cpp:87
```

Do you have any idea on how to solve this problem?

Thank you.

OpusK, 2020-05-12:
Hi, @munakano 

Would you please check the content of the link below?
It has not been updated since Agent 1.1.0. 
Therefore, it may not work.

munakano, 2020-05-15:
Thanks for getting back to me!
I tried in the following environment.
But it doesn't work.

**Environment**
Ubuntu 18.04.4 LTS
ROS2 Dashing Patch3
Micro-XRCE-DDS-Agent v1.1.0
ros2arduino 0.1.4

Actually, I don't use [Available boards](https://github.com/ROBOTIS-GIT/ros2arduino#available-boards-what-weve-tested-on-our-own-ram-size--32kb).
I use Feather M4 Express board, so it may not work.

Thank you for your help.

OpusK, 2020-05-19:
Hi, @munakano 

> Actually, I don't use Available boards.
I use Feather M4 Express board, so it may not work.

The boards listed in the link are just the boards I tested.
I think Adafruit's Feather M4 Express board will work.

Also, it seems that the error is not a problem with the board, but a problem due to mismatch of the version of Agent and rmw.

It is recommended to test the recommended version of the link below.
- [Recommend version](https://github.com/ROBOTIS-GIT/ros2arduino#version-specific-dependencies)

If I have time, I will test with the Feather M4 Express board.

closed 2020-05-27 by OpusK