ROBOTIS-GIT/ros2arduino #47: MicroXRCEAgent is not seem to work

james-yoo, 2020-06-17:
Hello @OpusK ,

I am working with ros2arduino using Arduino Due & ROS2 Dashing. I followed the instruction as below.
[image]

I run with this command after install publishing program to ESP32 with using ArduinoIDE v1.8.10,
 
 $ MicroXRCEAgent serial --dev /dev/ttyACM0 -b 115200

Result is as below:
[image]

And I cannot see topic(/arduino_chatter) from Arduino side.
Could you give me some advice to check?

Thanks

OpusK, 2020-06-18:
Hi, @james-yoo 

> $ MicroXRCEAgent serial --dev /dev/ttyACM0 -b 115200

Can you share detailed logs using the `-v(0~6)` option like below?

```bash
$ MicroXRCEAgent serial --dev /dev/ttyACM0 -b 115200 -v 6
```

james-yoo, 2020-06-18:
Hi, @OpusK 

I can see following with -v 6 option:
[image]

Is this what you asked for?

OpusK, 2020-06-19:
@james-yoo 

Are you uploading the basic examples without modification, and are you trying to communicate using USB?
If so, it will not work due to [different Serial instances of Arduino boards](https://www.arduino.cc/reference/en/language/functions/communication/serial/).
In other words, for DUE, you need to change it as below.
```cpp
#define XRCEDDS_PORT  SerialUSB
```

In the current example, there is no pre-processing for each board, so the user must modify it according to his board.
In this regard, for convenience, we will include preprocessing in the next release.

And I couldn't check because I didn't have DUE and ZERO. Could you please let me know if the USB connection is recognized as /dev/ttyACM* or /dev/ttyUSB*?

james-yoo, 2020-06-19:
@OpusK,

I have uploaded the basic example code without any modification and tried to use USB.
Currently, USB connection is showing /dev/ttyACM0 with Programming port on DUE.
Should I have to change to the Native USB port on DUE?

OpusK, 2020-06-19:
@james-yoo 

Sorry for the confusion.
I misunderstood.

Would you like to keep the port intact and press the reset button while the agent is running?

james-yoo, 2020-06-19:
@OpusK 

Thank you for your update. Please update if there is any progress on this issue.

james-yoo, 2020-07-20:
I would like to check if there is any further progress regarding the issue.

OpusK, 2020-07-21:
Hi, @james-yoo 

No progress has been made so far.
It seems to be quite time-consuming.

If there is any progress on this issue, I will let you know.
And since the operation is not perfect at this time, we will remove the DUE from the README.

amfern, 2020-11-03:
I am porting this library to INav, and i had a similar problem, this library would send packets over serial but the MicroXRCEAgent.
Switching to MicroXRCEAgent 1.3.2 resolved this issue

mjwhite8119, 2020-12-08:
Great work on this project James! 
I'm having the same issue documented here where I can successfully load the code onto the board and it connects to the Micro-ROS agent but does not publish the chatter topic.  The only thing is that the board that I'm loading onto is an ESP32.  I know you are not currently supporting that board but it's almost working and if you can point me in the right direction then I'd like to try and get it working for that board.  Please let me know what code segments to look at.  Thank you.