ROBOTIS-GIT/ros2arduino #52: not working on M5StickC & M5Stack  ("uxr_run_session_until_one_status" fails?)

ken551, 2020-07-23:
Hello.

I'm trying to execute `publisher_wifi_udp` example on my M5StickC & M5Stack, but it didn't work.

I use:
* Ubuntu 18.04
* MicroXRCE 1.3.0
* ROS2 Dashing

I only changed the definitions of SSID, SSID_PW, AGENT_IP, and topic name, but even that it didn't work on my environment.

When I type `ros2 node list` on Ubuntu PC, the node on M5StickC appears.
However, when I try `ros2 topic list`, the topic doesn't.

The log I got is as follows:
```
$ MicroXRCEAgent udp4 -p 8888 -v 6
Press CTRL+C to exit
[1595517735.034458] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1595517735.034679] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1595517744.003647] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA BB CC DD 81 00 FC 07
[1595517744.003931] info     | Root.cpp           | create_client            | create                 | client_key: 0xAABBCCDD, session_id: 0x81
[1595517744.004044] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x2864434397, address: 192.168.179.5:47138
[1595517744.004240] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1595517744.208223] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 104, data: 
0000: 81 80 00 00 01 05 5E 00 00 0A 00 11 01 02 00 00 50 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 72 6F 76 65 72 43 5F 70 75 62 5F 6E 6F 64
0040: 65 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63 69 70 61 6E 74 3E 3C 2F 64
0060: 64 73 3E 00 00 00 00 00
[1595517744.208293] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1595517744.212494] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0xAABBCCDD, participant_id: 0x001(1)
[1595517744.213185] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 11 00 00
[1595517744.213313] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1595517744.213432] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1595517744.241236] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1595517744.262316] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1595517744.262453] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1595517744.265069] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1595517744.265944] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 24, data: 
0000: 81 80 01 00 01 05 0F 00 00 0B 00 13 03 02 00 00 01 00 00 00 00 00 11 00
[1595517744.266141] debug    | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0xAABBCCDD, publisher_id: 0x001(3), participant_id: 0x001(1)
[1595517744.266236] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 13 00 00
[1595517744.266271] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1595517744.267676] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1595517744.269640] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1595517744.271013] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 24, data: 
0000: 81 80 02 00 01 05 0F 00 00 0C 00 14 04 02 00 00 01 00 00 00 00 00 11 00
[1595517744.271181] debug    | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0xAABBCCDD, subscriber_id: 0x001(4), participant_id: 0x001(1)
[1595517744.271288] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 02 00 05 01 06 00 00 0C 00 14 00 00
[1595517744.271319] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1595517744.275051] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1595517744.276067] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 120, data: 
0000: 81 80 03 00 01 03 6D 00 00 0D 00 52 02 02 00 00 5F 00 00 00 3C 64 64 73 3E 3C 74 6F 70 69 63 3E
0020: 3C 6E 61 6D 65 3E 53 74 72 69 6E 67 3C 2F 6E 61 6D 65 3E 3C 64 61 74 61 54 79 70 65 3E 73 74 64
0040: 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 53 74 72 69 6E 67 5F 3C 2F 64 61 74 61 54
0060: 79 70 65 3E 3C 2F 74 6F 70 69 63 3E 3C 2F 64 64 73 3E 00 00 11 00 00 00
[1595517744.276281] debug    | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xAABBCCDD, topic_id: 0x005(2), participant_id: 0x001(1)
[1595517744.276370] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 03 00 05 01 06 00 00 0D 00 52 00 00
[1595517744.276400] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1595517744.279536] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1595517744.280467] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 168, data: 
0000: 81 80 04 00 01 05 9F 00 00 0E 00 15 05 02 00 00 91 00 00 00 3C 64 64 73 3E 3C 64 61 74 61 5F 77
0020: 72 69 74 65 72 3E 3C 74 6F 70 69 63 3E 3C 6B 69 6E 64 3E 4E 4F 5F 4B 45 59 3C 2F 6B 69 6E 64 3E
0040: 3C 6E 61 6D 65 3E 72 74 2F 63 68 61 74 74 65 72 3C 2F 6E 61 6D 65 3E 3C 64 61 74 61 54 79 70 65
0060: 3E 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 53 74 72 69 6E 67 5F 3C 2F 64
0080: 61 74 61 54 79 70 65 3E 3C 2F 74 6F 70 69 63 3E 3C 2F 64 61 74 61 5F 77 72 69 74 65 72 3E 3C 2F
00A0: 64 64 73 3E 00 00 13 00
[*** LOG ERROR #0001 ***] [2020-07-24 00:22:24] [] {argument index out of range}
[1595517744.280792] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 04 00 05 01 06 00 00 0E 00 15 80 00
[1595517744.280817] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 05 00 00 00 80
[1595517744.283845] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 05 00 00 00 80
^C[1595517745.642334] info     | UDPv4AgentLinux.cpp | fini                     | server stopped         | port: 8888

```

And, I found that the `uxr_run_sesson_until_one_status` method (`session.c` L268) called from `new ros2::Publisher...`(`ros2.hpp` L68) doesn't seem to return `true`, and then `p_pub` is deleted and nullptr is returned.
I'm not sure what code is returned, but at least `p_pub->is_registered` returns false.

Could you tell me if there's any solution to that? Thank you.

OpusK, 2020-07-24:
Hi, @ken551 

This is the result of a failure to create a data_writer and data_reader.

> [*** LOG ERROR #0001 ***] [2020-07-24 00:22:24] [] {argument index out of range}

If you look at this error, it says that the argument is over length.
Among the changes you made is the topic name, what did you change?

Please check if the same error message is displayed even with no modification.

ken551, 2020-07-24:
@OpusK Hello, thank you for checking.

I changed the topic name from `arduino_chatter` to `chatter`.
The same error message also appears even when I don't change anything.

Also, I downgraded the ros2arduino to `v1.4.0` and microXRCE to `v1.1.0`, and both pub and sub work perfectly.

OpusK, 2020-07-24:
Was your ROS2 dashing version patch6?
MicroXRCE Agent depends on the version of ROS2 (the version of FastRTPS).

ken551, 2020-07-24:
@OpusK 
I checked the version with `sudo apt-cache policy ros-dashing-desktop` (I've never tried this before... does this work properly?), and it said `0.7.3(2020512)` was installed.
So, I gonna downgrade my ROS2 dashing to version `0.6.*` and try again.

I'll comment here again when the situation changes.
Thank you!

OpusK, 2020-07-24:
> I checked the version with sudo apt-cache policy ros-dashing-desktop (I've never tried this before... does this work properly?), and it said 0.7.3(2020512) was installed.

I'm not sure because it's different from the patch version naming rules.
However, unlike Patch6 was released in March, it is in May.

Anyway, thanks for sharing :)

closed 2020-07-24 by OpusK