ROBOTIS-GIT/ros2arduino #53: error: size of array 'data' is too large by using UNO

JuFengWu, 2020-09-09:
Hello:

I use Arduino UNO and it used Atmega328, so it has 32KB ram size.

When I compile the example program, it shows 

Arduino: 1.8.12 (Linux), Board: "Arduino Uno"

In file included from /home/leowu/Arduino/libraries/ros2arduino/src/uxr/client/core/type/xrce_types.h:1:0,
                 from /home/leowu/Arduino/libraries/ros2arduino/src/ros2/xrcedds/micro_xrce_dds/lib/include/uxr/client/core/session/session.h:25,
                 from /home/leowu/Arduino/libraries/ros2arduino/src/uxr/client/core/session/session.h:1,
                 from /home/leowu/Arduino/libraries/ros2arduino/src/ros2/xrcedds/micro_xrce_dds/lib/include/uxr/client/client.h:24,
                 from /home/leowu/Arduino/libraries/ros2arduino/src/uxr/client/client.h:1,
                 from /home/leowu/Arduino/libraries/ros2arduino/src/ros2/xrcedds/micro_xrce_dds/micro_xrce_dds.h:18,
                 from /home/leowu/Arduino/libraries/ros2arduino/src/ros2/xrcedds/xrcedds.hpp:12,
                 from /home/leowu/Arduino/libraries/ros2arduino/src/ros2/publisher.hpp:12,
                 from /home/leowu/Arduino/libraries/ros2arduino/src/ros2/ros2.hpp:12,
                 from /home/leowu/Arduino/libraries/ros2arduino/src/ros2arduino.h:15,
                 from /home/leowu/hosekilu_source_code/arduino/test_ardunio_ros2/test_ardunio_ros2.ino:1:
/home/leowu/Arduino/libraries/ros2arduino/src/ros2/xrcedds/micro_xrce_dds/lib/include/uxr/client/core/type/xrce_types.h:819:55: error: size of array 'data' is too large
     PackedSamples data[UXR_PACKED_SAMPLES_SEQUENCE_MAX];
                                                       
In file included from /home/leowu/Arduino/libraries/ros2arduino/src/ros2/publisher.hpp:12:0,
                 from /home/leowu/Arduino/libraries/ros2arduino/src/ros2/ros2.hpp:12,
                 from /home/leowu/Arduino/libraries/ros2arduino/src/ros2arduino.h:15,
                 from /home/leowu/hosekilu_source_code/arduino/test_ardunio_ros2/test_ardunio_ros2.ino:1:
/home/leowu/Arduino/libraries/ros2arduino/src/ros2/xrcedds/xrcedds.hpp:103:66: warning: large integer implicitly truncated to unsigned type [-Woverflow]
 void init(uint8_t rtps_product,unsigned int client_key=0xAABBCCDD);

                                                                  
In file included from /home/leowu/Arduino/libraries/ros2arduino/src/ros2arduino.h:15:0,
                 from /home/leowu/hosekilu_source_code/arduino/test_ardunio_ros2/test_ardunio_ros2.ino:1:
/home/leowu/Arduino/libraries/ros2arduino/src/ros2/ros2.hpp:27:85: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     Node(const char* node_name = "ros2_xrcedds_participant",unsigned int client_key=0xAABBCCDD);

                                                                                  
/home/leowu/Arduino/libraries/ros2arduino/src/ros2/ros2.hpp:32:94: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     void recreate(const char* node_name = "ros2_xrcedds_participant",unsigned int client_key=0xAABBCCDD);

                                                                                              
/home/leowu/hosekilu_source_code/arduino/test_ardunio_ros2/test_ardunio_ros2.ino: In constructor 'StringPub::StringPub()':
/home/leowu/hosekilu_source_code/arduino/test_ardunio_ros2/test_ardunio_ros2.ino:19:32: warning: large integer implicitly truncated to unsigned type [-Woverflow]
   : Node("ros2arduino_pub_node")
                                
/home/leowu/hosekilu_source_code/arduino/test_ardunio_ros2/test_ardunio_ros2.ino: In constructor 'LedSub::LedSub()':
/home/leowu/hosekilu_source_code/arduino/test_ardunio_ros2/test_ardunio_ros2.ino:37:32: warning: large integer implicitly truncated to unsigned type [-Woverflow]
   : Node("ros2arduino_sub_node")
                                
exit status 1
Error compiling for board Arduino Uno.

It seems the problem is UXR_PACKED_SAMPLES_SEQUENCE_MAX size too big.

Is this a issue?  or I should make the ram size bigger?

OpusK, 2020-09-11:
Hi, @JuFengWu 

ros2arduino does not support UNO/Mega.
Please refer to [README](https://github.com/ROBOTIS-GIT/ros2arduino#restrictions) for more detail.

JuFengWu, 2020-09-12:
Thank you!
I found UNO has 32kb RAM, but some memory is used for boot loader.

closed 2020-09-12 by JuFengWu