ROBOTIS-GIT/ros2arduino #55: Agent does not recognize

kakank4444, 2020-11-21:
I want to use ros2arduino
I confirmed the current version of ros2
```
$ apt-cache show ros-dashing-desktop

Package: ros-dashing-desktop
Version: 0.7.4-1bionic.20201024.022440
Architecture: amd64
```

I noticed that it doesn't match the recommended version

I don't know some patches, so I want to install "Dashing Diademata Patch 6", but I don't know the procedure.
Is the wget URL changed in the https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Development-S

kakank4444, 2020-12-03:
Hello @OpusK 

Successful installation of Patch 6 from previous post

Although the node is recognized
I don't recognize the topic.

```
$ ros2 node list
/ros2arduino_pub_node
```

```
$ ros2 topic list
/parameter_events
/rosout
```

ArduinoIDE: 1.8.14
Hard: Esp32 Devkitc
OS: Ubuntu18.04
ROS2: Dashing Diademata Patch6

ROS2 built from source code
Basically installed according to the following

ros2arduino | ROS2 | Micro-XRCE-DDS Agent
-- | -- | --
0.2.1 | Dashing Diademata Patch6 | 1.3.0

```
$ git clone https://github.com/eProsima/Micro-XRCE-DDS-Agent.git
$ cd Micro-XRCE-DDS-Agent && git checkout v1.3.0
$ mkdir build && cd build
$ source /opt/ros/dashing/setup.bash # to share libraries with ros2
$ cmake ..
$ make
$ sudo make install
$ sudo ldconfig /usr/local/lib/
```
but

`Source /opt/ros/dashing/setup.bash`

To

`Source "ros2 install target"/install/setup.bash`

Changed to
The source code uses the sample "publisher" as it is.

Paste the log below. Isn't topic generation going well?
```
sudo MicroXRCEAgent serial --dev /dev/ttyUSB0 -b 115200 -v 6
Press CTRL+C to exit
[1606963842.553090] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1606963842.553489] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1606963852.467867] info     | Root.cpp           | create_client            | create                 | client_key: 0xAABBCCDD, session_id: 0x81
[1606963852.467990] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x2864434397, address: 0
[1606963852.468228] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1606963852.472576] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA BB CC DD 81 00 FC 07
[1606963852.472690] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x2864434397, address: 0
[1606963852.472819] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1606963852.478380] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA BB CC DD 81 00 FC 07
[1606963852.478465] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x2864434397, address: 0
[1606963852.478552] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 19, data: 
[... 90 lines cut ...]
[1606963853.276474] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1606963853.350381] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 176, data: 
0000: 81 80 04 00 01 05 A7 00 00 0E 00 15 05 02 00 00 99 00 00 00 3C 64 64 73 3E 3C 64 61 74 61 5F 77
0020: 72 69 74 65 72 3E 3C 74 6F 70 69 63 3E 3C 6B 69 6E 64 3E 4E 4F 5F 4B 45 59 3C 2F 6B 69 6E 64 3E
0040: 3C 6E 61 6D 65 3E 72 74 2F 61 72 64 75 69 6E 6F 5F 63 68 61 74 74 65 72 3C 2F 6E 61 6D 65 3E 3C
0060: 64 61 74 61 54 79 70 65 3E 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 53 74
0080: 72 69 6E 67 5F 3C 2F 64 61 74 61 54 79 70 65 3E 3C 2F 74 6F 70 69 63 3E 3C 2F 64 61 74 61 5F 77
00A0: 72 69 74 65 72 3E 3C 2F 64 64 73 3E 00 00 13 00
[*** LOG ERROR ***] [2020-12-03 11:50:53] [] argument index out of range
[1606963853.350796] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 14, data: 
0000: 81 80 04 00 05 01 06 00 00 0E 00 15 80 00
[1606963853.350821] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 05 00 00 00 80
[1606963853.351433] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 04 00 04 00 80
[1606963853.351549] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 05 00 00 00 80
[1606963853.361250] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 04 00 04 00 80
[1606963853.364515] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 05 00 00 00 80
[1606963853.365390] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0xAABBCCDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 05 00 00 00 80
```

ROBOTIS-Will, 2020-12-03:
Hi @kakank4444 
Looks like the publisher has not been successfully created, and therefore, causing the argument index error.
Although we have somewhat developed package to run ROS2 on Arduino, recently eProsima(Fast RTPS developer) is putting a lot of effort on supporting ROS2 on embedded boards including Arduino boards.
https://github.com/micro-ROS/micro_ros_arduino
I'd recommend try their most up to date package instead.
Thank you.

kakank4444, 2020-12-04:
@ROBOTIS-Will 
Thanks, if this problem doesn't go away I'll give it a try.

@OpusK 
I would like to use "ros2arduino" if possible, so please let me know.
Is it related to the version of FastRTPS that Publisher can't generate?

robertkovax, 2023-02-15:
I'm having the similar issue with my Arduino RP2040 Connect and Nicla Vision boards. Node is listed but the topic is not appearing. Any ideas?
log:
[*** LOG ERROR #0001 ***] [2023-02-15 16:08:02] [] {argument not found}

lagessiehcs, 2024-03-11:
> Hello @OpusK
>
> Successful installation of Patch 6 from previous post
>
> Although the node is recognized I don't recognize the topic.
>
> ```
> $ ros2 node list
> /ros2arduino_pub_node
> ```
>
> ```
> $ ros2 topic [...]

How did you install Patch6?