ROBOTIS-GIT/ros2arduino #56: ESP32/ Arduino Uno: error: size of array 'data' is too large

tom-garford, 2020-11-27:
Hi All, 

Just a quick one. I am trying to compile the simple serial publisher in ros2arduino (version 0.2.1) with ROS2 dashing. I can successfully install and run the MicroXRCEAgent on my laptop however, am having trouble compiling the publisher example through Arduino IDE for an ESP32-based board with 32kb of flash memory. 

The error is: 
```
In file included from /home/tom/Arduino/libraries/ros2arduino/src/uxr/client/core/type/xrce_types.h:1:0,
                 from /home/tom/Arduino/libraries/ros2arduino/src/ros2/xrcedds/micro_xrce_dds/lib/include/uxr/client/core/session/session.h:25,
                 from /home/tom/Arduino/libraries/ros2arduino/src/uxr/client/core/session/session.h:1,
                 from /home/tom/Arduino/libraries/ros2arduino/src/ros2/xrcedds/micro_xrce_dds/lib/include/uxr/client/client.h:24,
                 from /home/tom/Arduino/libraries/ros2arduino/src/uxr/client/client.h:1,
                 from /home/tom/Arduino/libraries/ros2arduino/src/ros2/xrcedds/micro_xrce_dds/micro_xrce_dds.h:18,
                 from /home/tom/Arduino/libraries/ros2arduino/src/ros2/xrcedds/xrcedds.hpp:12,
                 from /home/tom/Arduino/libraries/ros2arduino/src/ros2/publisher.hpp:12,
                 from /home/tom/Arduino/libraries/ros2arduino/src/ros2/ros2.hpp:12,
                 from /home/tom/Arduino/libraries/ros2arduino/src/ros2arduino.h:15,
                 from /home/tom/Arduino/libraries/ros2arduino/examples/publisher/publisher.ino:2:
/home/tom/Arduino/libraries/ros2arduino/src/ros2/xrcedds/micro_xrce_dds/lib/include/uxr/client/core/type/xrce_types.h:819:55: error: size of array 'data' is too large
     PackedSamples data[UXR_PACKED_SAMPLES_SEQUENCE_MAX];
                                                       ^
```
The README says >=32kb RAM, but can someone confirm this works on an Uno? If not, I would be happy to change boards to a MKR Zero if that is more recommended. 

Thanks, Tom.

tom-garford, 2020-11-27:
Silly question, my board only has 2k of RAM but 32kb of flash memory..

closed 2020-11-27 by tom-garford

kappa95, 2021-04-12:
> Silly question, my board only has 2k of RAM but 32kb of flash memory..

I did the same question, you provide the answer :+1: 
Thank you!