ROBOTIS-GIT/ros2arduino #62: Portenta H7 with Jetson NX

guilleherreraf, 2021-02-11:
Hello!
I'm working on a project where I use a NVIDIA Jetson NX (with Ubuntu 18 and ROS Dashing) as a high-level controller, and I want to send ROS messages to a Portenta H7 acting as a low-level controller. I uploaded the `publisher.ino` example to the Portenta and installed the Micro-XRCE-DDS Agent on the NVIDIA NX. Afterwards, I connected the Portenta via USB to the NX and launched the agent: `./MicroXRCEAgent serial --dev /dev/ttyACM0 -b 115200`. However, when I run `ros2 topic list`, I'm not able to see the `/arduino_chatter` topic. 
I also tried checking if there is some data being sent to the USB port, with `screen`, but there is nothing being sent to the port. Any idea what might be happening?
Thanks in advance!

ROBOTIS-Will, 2021-02-25:
Hi @guilleherreraf 
I'm sorry about the delayed response.
Looks like you've already found the micro ros arduino which will be replacing ros2arduino.
Linking the thread of micro ros arduino below for other developers.
https://github.com/micro-ROS/micro_ros_arduino/issues/111
Thanks.

guilleherreraf, 2021-02-25:
Hello @ROBOTIS-Will, I managed to make it work via Ethernet (UDP). I gave micro-ROS a try, but I could not make it work. For my implementation I will keep using the `ros2arudino` package.