ROBOTIS-GIT/ros2arduino #63: Error verifying code in arduino

azzopardim, 2021-02-13:
In file included from C:\Users\micha\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.11\cores\arduino/SafeRingBuffer.h:25:0,
                 from C:\Users\micha\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.11\cores\arduino/Uart.h:23,
                 from C:\Users\micha\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.11\variants\arduino_zero/variant.h:43,
                 from C:\Users\micha\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.11\cores\arduino/Arduino.h:48,
                 from sketch\publisher.ino.cpp:1:
C:\Users\micha\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.11\cores\arduino/sync.h:28:22: error: expected unqualified-id before 'for'
 #define synchronized for (__Guard __guard; __guard.enter(); )
                      ^
D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/ros2/xrcedds/micro_xrce_dds/lib/include/uxr/client/core/session/session.h:97:10: note: in expansion of macro 'synchronized'
     bool synchronized;
          ^~~~~~~~~~~~
C:\Users\micha\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.11\cores\arduino/sync.h:28:44: error: '__guard' does not name a type; did you mean '__Guard'?
 #define synchronized for (__Guard __guard; __guard.enter(); )
                                            ^
D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/ros2/xrcedds/micro_xrce_dds/lib/include/uxr/client/core/session/session.h:97:10: note: in expansion of macro 'synchronized'
     bool synchronized;
          ^~~~~~~~~~~~
C:\Users\micha\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.11\cores\arduino/sync.h:28:61: error: expected unqualified-id before ')' token
 #define synchronized for (__Guard __guard; __guard.enter(); )
                                                             ^
D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/ros2/xrcedds/micro_xrce_dds/lib/include/uxr/client/core/session/session.h:97:10: note: in expansion of macro 'synchronized'
     bool synchronized;
          ^~~~~~~~~~~~
exit status 1
Error compiling for board Arduino Zero (Native USB Port).

azzopardim, 2021-02-13:
I replaced the library downloaded from the IDE with the one on GitHub and also got several issues:#

In file included from D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/uxr/client/config.h:18:0,
                 from D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/ros2/xrcedds/micro_xrce_dds/micro_xrce_dds_xml.h:12,
                 from D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/ros2/xrcedds/micro_xrce_dds/micro_xrce_dds.h:17,
                 from D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/ros2/xrcedds/xrcedds.hpp:12,
                 from D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/ros2/publisher.hpp:12,
                 from D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/ros2/ros2.hpp:12,
                 from D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/ros2arduino.h:15,
                 from D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\examples\publisher\publisher.ino:2:
D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/user_config.h:23:0: warning: "USER_ROS2_PUBLISHER_MAX" redefined
 #define USER_ROS2_PUBLISHER_MAX  10
 
In file included from D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\examples\publisher\publisher.ino:1:0:
sketch\user_config.h:21:0: note: this is the location of the previous definition
 #define USER_ROS2_PUBLISHER_MAX  5
 
In file included from D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/uxr/client/config.h:18:0,
                 from D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/ros2/xrcedds/micro_xrce_dds/micro_xrce_dds_xml.h:12,
                 from D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/ros2/xrcedds/micro_xrce_dds/micro_xrce_dds.h:17,
                 from D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/ros2/xrcedds/xrcedds.hpp:12,
                 from D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/ros2/publisher.hpp:12,
                 from D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/ros2/ros2.hpp:12,
                 from D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/ros2arduino.h:15,
                 from D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\examples\publisher\publisher.ino:2:
D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/user_config.h:24:0: warning: "USER_ROS2_SUBSCRIBER_MAX" redefined
 #define USER_ROS2_SUBSCRIBER_MAX 10
 
In file included from D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\examples\publisher\publisher.ino:1:0:
sketch\user_config.h:22:0: note: this is the location of the previous definition
 #define USER_ROS2_SUBSCRIBER_MAX 5
 
In file included from C:\Users\micha\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.11\cores\arduino/SafeRingBuffer.h:25:0,
                 from C:\Users\micha\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.11\cores\arduino/Uart.h:23,
                 from C:\Users\micha\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.11\variants\arduino_zero/variant.h:43,
                 from C:\Users\micha\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.11\cores\arduino/Arduino.h:48,
                 from sketch\publisher.ino.cpp:1:
C:\Users\micha\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.11\cores\arduino/sync.h:28:22: error: expected unqualified-id before 'for'
 #define synchronized for (__Guard __guard; __guard.enter(); )
                      ^
D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/ros2/xrcedds/micro_xrce_dds/lib/include/uxr/client/core/session/session.h:97:10: note: in expansion of macro 'synchronized'
     bool synchronized;
          ^~~~~~~~~~~~
C:\Users\micha\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.11\cores\arduino/sync.h:28:44: error: '__guard' does not name a type; did you mean '__Guard'?
 #define synchronized for (__Guard __guard; __guard.enter(); )
                                            ^
D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/ros2/xrcedds/micro_xrce_dds/lib/include/uxr/client/core/session/session.h:97:10: note: in expansion of macro 'synchronized'
     bool synchronized;
          ^~~~~~~~~~~~
C:\Users\micha\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.11\cores\arduino/sync.h:28:61: error: expected unqualified-id before ')' token
 #define synchronized for (__Guard __guard; __guard.enter(); )
                                                             ^
D:\Users\micha\OneDrive\Documents\Arduino\libraries\ros2arduino\src/ros2/xrcedds/micro_xrce_dds/lib/include/uxr/client/core/session/session.h:97:10: note: in expansion of macro 'synchronized'
     bool synchronized;
          ^~~~~~~~~~~~
exit status 1
Error compiling for board Arduino Zero (Native USB Port).

jsamwel, 2021-02-28:
There is an issue between the latest version of the Arduino SAMD boards library and the ros2arduino library. Which is being worked on as seen in this issue.
https://github.com/arduino/ArduinoCore-samd/issues/596

But if you want your code working now you can revert to an older version of the Arduino SAMD boards. Version 1.8.10 seems to be working fine with the ros2arduino library. Which is also discussed in this thread on the arduino forum
https://forum.arduino.cc/index.php?topic=724140.0

closed 2021-06-09 by azzopardim