ROBOTIS-GIT/ros2arduino #65: ros2arduino.refs error

Kakcalu13, 2021-04-13:
```
bwuk@robots:~/Micro-XRCE-DDS-Agent/build$ ls -l /dev/ttyACM0
crw-rw-rw- 1 root dialout 166, 0 Apr 12 18:04 /dev/ttyACM0
bwuk@robots:~/Micro-XRCE-DDS-Agent/build$ MicroXRCEAgent serial --dev /dev/ttyACM0 -b 115200 -r ros2arduino.refs
[1618272496.596805] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
2021-04-12 18:08:16.597 [XMLPARSER Error] Invalid element found into 'builtinAttributesType'. Name: domainId -> Function getXMLBuiltinAttributes
2021-04-12 18:08:16.597 [XMLPARSER Error] Error parsing participant profile -> Function parseXMLParticipantProf
2021-04-12 18:08:16.597 [XMLPARSER Error] Error parsing profile's tag participant -> Function parseProfiles
2021-04-12 18:08:16.597 [XMLPARSER Error] Error parsing profile's tag data_writer -> Function parseProfiles
2021-04-12 18:08:16.597 [XMLPARSER Error] Error parsing profile's tag data_reader -> Function parseProfiles
2021-04-12 18:08:16.597 [XMLPARSER Error] Error parsing profile's tag topic -> Function parseProfiles
2021-04-12 18:08:16.597 [XMLPARSER Error] Error parsing profile's tag topic -> Function parseProfiles
2021-04-12 18:08:16.597 [XMLPARSER Error] Error parsing 'ros2arduino.refs' -> Function loadXMLFile
[1618272496.597487] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
^C
bwuk@robots:~/Micro-XRCE-DDS-Agent/build$ 
```

my ros2arduino.refs file:
```
 <profiles>
 	<participant profile_name="ros2arduino_xml_node">
 		<rtps>
 			<name>ros2arduino_basic_node</name>
 			<builtin>
 				<domainId>0</domainId>
 			</builtin>
 		</rtps>
 	</participant>

 	<data_writer profile_name="arduino_chatter">
 		<topic>
 			<kind>NO_KEY</kind>
 			<name>rt/arduino_chatter</name>
 			<dataType>std_msgs::msg::dds_::String_</dataType>
 			<historyQos>
 				<kind>KEEP_LAST</kind>
 				<depth>10</depth>
 			</historyQos>
 		</topic>
 	</data_writer>

 	<data_reader profile_name="arduino_led">
 		<topic>
 			<name>rt/arduino_led</name>
 			<dataType>std_msgs::msg::dds_::Bool_</dataType>
 		</topic>
 	</data_reader>

 	<topic profile_name="Bool">
 		<kind>NO_KEY</kind>
 		<name>Bool</name>
 		<dataType>std_msgs::msg::dds_::Bool_</dataType>
 	</topic>

 	<topic profile_name="String">
 		<kind>NO_KEY</kind>
 		<name>String</name>
 		<dataType>std_msgs::msg::dds_::String_</dataType>
 	</topic>
 </profiles>
```

Was I supposed to put something in <kind>? It's saying NO_KEY. 

I followed the steps over and over. It seems I get the same result.

OS: 20.04 LTS Focal
Ros2: Foxy
Ardiuno: Ardiuno 1.8.14 Hourly Build 2021
Laptop: Dell xps 13
CPU: I7

Any idea why is this happening?

syafiqsallehudin, 2021-08-05:
Hi, I am having almost the same issue. Any progress on this issue?

Kakcalu13, 2021-08-05:
> Hi, I am having almost the same issue. Any progress on this issue?

I end up using Micro-ROS. You should try that one out! I'm closing this since it's no longer issue for me.

closed 2021-08-05 by Kakcalu13