ROBOTIS-GIT/ros2arduino #72: Arduino UNO compiling issue

leonicode, 2022-10-30:
Hi to all,
I installed the library 0.2.1 in my Arduino IDE 2.0.0 but I've a message error when I verify the publisher.ino sketch:

xrce_types.h:819:55: error: size of array 'data' is too large
     PackedSamples data[UXR_PACKED_SAMPLES_SEQUENCE_MAX];

Is this library compatible with Arduino UNO (it has 32kb of RAM) ?
Thanks
Leonard

Htet-Wai-Yan-HWY, 2023-09-16:
I have same issues with this arduino  uno. 
please let me know if you solved this problem.  Thank.

rehmanazeem, 2023-10-28:
Has somebody found a solution to this issue? I am facing the same compiling error on Arduino Uno.
Any leads would be helpful. TIA

Htet-Wai-Yan-HWY, 2023-10-29:
I used micro_ros_arduino library instead of ros2arduino because micro_ros_arduino have a lot of community and example guides. I recommend to use micro_ros_arduino.

jackhou66, 2024-03-17:
me encounter the same problem

jackhou66, 2024-03-17:
> I used micro_ros_arduino library instead of ros2arduino because micro_ros_arduino have a lot of community and example guides. I recommend to use micro_ros_arduino.

so ,you use  micro_ros_arduino  on your  arduino uno board?

Htet-Wai-Yan-HWY, 2024-03-17:
my apologies, I didn't see the uno board. Can't use micro_ros_arduino in uno board.It have to have SOC chip to use micro_ros_arduino.

jackhou66, 2024-03-17:
It seems can not  support uno r3 . It has definite kinds of board.

duy621, 2024-04-04:
Did anyone solved this problem ? Now I face the same problem, are there any leads for this ? 
TIA

jackhou66, 2024-04-04:
Which type of device. And which mediaware  you use?

duy621, 2024-04-04:
@jackhou66 I used the arduino UNO with the arduino IDE 2. I want to create a subscriber node in arduino sketch to subscribe to a message (float32 data type) from a ros2 node. Now I kinda stuck at this point with this compilation error now.

jackhou66, 2024-04-04:
Uno version is 3?haven't WIFI?

duy621, 2024-04-04:
@jackhou66 It is the UNO R3, without WIFI

jackhou66, 2024-04-04:
not supported device, ,,, you can check the supported list on their website
..

duy621, 2024-04-04:
@jackhou66 thanks for letting me know. But however I find it weird as I found on the ros2arduino documentation that they state: "This library is compatible with all architectures so you should be able to use it on all the Arduino boards." :D But anyway I might have to change solution tho

moun20, 2024-05-19:
@jackhou66 hi .. I am working with the ros2 humble and arduino uno R3 my question is . have you successfully communicate ros2 with arduino uno using ros2arduino  let me Know plz .. and if you are funding another solution too .. thanks

duy621, 2024-05-19:
@moun20 i have tried and I can confirm that it would not work with ros2 humble and arduino. The arduino board R3 is simply not powerful enough in terms of reaource to support for the ROS data communication. 
I am currently on the same project with ROS and Arduino, my advice is to go for another microcontroller board with more powerful hardware (STM32 for example)
If you still want to use arduino R3, go back to use ros1 Noetic. That is what im currently using in my project with arduino R3. Note that with only 2kg of SRAM might limit you on having global variables and transferring ros messages
KG

jackhou66, 2024-05-19:
You can try esp32 as Middleware  between jetson nano and arduino