ROBOTIS-GIT/ros2arduino #78: No nodes or topics are displayed.

tuananh0606-tech, 2024-07-10:
Hi, I tried running the publisher example with ros2 humble and esp32 using xrce dds they connected but no new topics or nodes were created. Help me, please!
[image]
[image]
[image]

Vinothhk, 2024-08-27:
Did you resolve this issue

tuananh0606-tech, 2024-08-28:
@Vinothhk I don't, but I think because I installed XRCE-DDS before, there was a conflict, I couldn't completely remove XRCE-DDS